gregtomasch/EM7180_SENtral_Calibration #3: MAG readings saturate at -1000 uT instead of -4915 uT (Teensy_ST_USFS)

rousselmanu, 2019-05-28:
Hi,
I am using the ST board from Pesky Products (EM7180+LSM6DSM+LIS2MDL).
The USFS board is near a quite strong steady field, and the Z axis is saturated at -1000 uT.
According to the code in "config.h", the magnetic sensor output scale should be +/-4915 uT.
Is there something I can do to use a larger scale on the magnetometers ?
Thank you for your feedback.

rousselmanu, 2019-05-28:
I just noticed that 4915 uT is 0x1333 in hexadecimal and not 0x133, so [line 107](https://github.com/gregtomasch/EM7180_SENtral_Calibration/blob/master/Teensy_ST_USFS_Baseline_Calibration_Utility_Acc_WS/config.h#L107) should be corrected.
But the behavior is still the same: saturation at -1000 uT.

gregtomasch, 2019-05-28:
Hello,

Line 107 is correct; this quantity is not directly in physical units but instead it is the number that the Sentral driver needs to properly scale the magnetometer data internally to deliver output at resolution of 1.0f = 50uT. One of the quirks of the Sentral is that it always delivers data at a fixed resolution/scale factor, no matter what sensor range is selected:

Rotational velocity: 1.0f = 1Rad/s
Acceleration: 1.0f  = 1g
Magnetic field: 1.0f = 50uT

The LIS2MDL magnetometer only has one range and the scale factor is set so that the indicated from the Sentral is physically correct...

I would suggest that the solution to your practical problem is address the enormous stray field...

rousselmanu, 2019-05-29:
Thank you for your answer.
So if I understand well the LIS2MDL sensor range is +/-4915 uT (fixed), but the Sentral scaled magnetometer output range is only +/-1000 uT. Is that right? Does it still use the full sensor scale internally?
I noticed that if I ask for unscaled sensor data from the Sentral (value 0x02 sent to "AlgorithmControl" register), the Z output is around -800 uT instead of being saturated at -1000.

I cannot reduce this steady magnetic field due to my specific application, and I thought that it could be compensated by the autocalibration algorithm as a soft-iron distortion because it is still within +/-4915 uT...
The temporary solution I found is to use the 6DOF mode instead of 9DOF by setting AlgorithmControl to 0x48.

gregtomasch, 2019-05-29:
When the unscaled sensor data option is selected, ALL scaling and orientation matrix actions are eliminated and the data stream reverts to what is provided by the sensor alone. So yes, the scaling (and sensing axis orientation) will change but you need to refer to the LIS2MDL datasheet to determine what the differences would be...

These relatively minor points aside, the reality is that the geomagnetic field is roughly on the order of 50uT and you are trying to measure its components with an interfering field of ~1000uT. I strongly doubt this would be successful no matter what fusion algorithm is used...

The reference frame of the distorting field is important as well. If it is static with respect to the magnetometer ( i.e. body frame) it will appear as a HARD IRON distortion (magnetometer response surface translated from the origin). In this case, the stray field can in principle be compensated but it is well beyond what the SpacePoint algorithm can do unaided. If the stray field is static with respect to world coordinates, it simply cannot be corrected...

rousselmanu, 2019-05-29:
Yes, the distorting field is static with respect to the magnetometer, so as you say it is basically a (big!) offset to the magnetometer readings. But I understand that it is probably way too big to be compensated. 
All in all, I see that the magnetometers cannot really be of any help to the data fusion algorithms in my application. I will just use the 6DOF version.

And to conclude on the initial saturation problem, from the EM7180 datasheet p. 15: "When the
output is scaled, the magnetometer data is scaled to ±1000 µT". So even if the magnetometer itself has a larger scale. Maybe could you specify it on the main page under "sensor scales"?

Thank you very much for your time and help, and for your teaching efforts.

closed 2019-05-29 by rousselmanu