gregtomasch/EM7180_SENtral_Calibration #6: Slightly misleading instructions for calibration

DavidAntliff, 2019-09-18:
Suggest changing "As an initial check, hold the USFS level and right side up" to "As an initial check, hold the USFS level with the top side facing up".

https://github.com/gregtomasch/EM7180_SENtral_Calibration/blame/master/README.md#L123

I suggest this because I spent 10 minutes with it on its left edge ("right side up"!), wondering how having the right-hand side of the board vertically oriented was related to an upside down board! Silly, I know, but I was misled for a time.

gregtomasch, 2019-10-07:
Really?

DavidAntliff, 2019-10-08:
Yes, really. Clarity is important, don't you think? The word "right side" is not a good alternative to "top side", in my opinion, when talking about orientation. I wonder if "right side up" is an idiomatic Americanism? I'm familiar with it in the context of cardboard boxes, but it isn't the first thing that comes to mind when trying to orient an IMU.

closed 2020-12-21 by gregtomasch