gregtomasch/EM7180_SENtral_Calibration #7: Quaternions point to true north, YPR points to magnetic north

DavidAntliff, 2019-09-19:
In case this helps anyone in future - the output values from the sketch differ in that the Quaternion orientation includes the magnetic declination, therefore orienting relative to true north, whereas the yaw, pitch, roll output is relative to magnetic north and does not include the magnetic declination.

For example, in my location with a magnetic declination of about +22 degrees 40 minutes (positive, east) I had to set `MAG_DECLINIATION -22.67` (note the negative sign). Then when the board's axis is aligned with true north:

```
Sentral_0 Quaternion (NED):
Q0_0 = 1.00 Qx_0 = 0.01 Qy_0 = 0.01 Qz_0 = -0.01

Sentral_0 Yaw, Pitch, Roll: 336.34, 0.92, 1.43
```

You can see that the quaternion is very close to the identity quaternion (1, 0, 0, 0), but the Yaw reading is out by about 24 degrees.

When aligned with magnetic north:

```
Sentral_0 Quaternion (NED):
Q0_0 = 0.98 Qx_0 = 0.01 Qy_0 = 0.01 Qz_0 = 0.19

Sentral_0 Yaw, Pitch, Roll: 359.08, 0.49, 1.34
```

In this case, the Yaw reading is almost due north, however the quaternion translates to a yaw of about 22 degrees.

If this is intentional then this isn't an issue and can be closed, however may I suggest that it could be documented in the main guide for future reference?

gregtomasch, 2019-10-07:
Magnetic declination is not handled inside the EM7180. Whether the output is selected as quaternions or Euler angles, local magnetic declination needs to be handled on the host MCU...

DavidAntliff, 2019-10-08:
I may have got it backwards: the quaternions, read from the IMU, will indicate magnetic north, and the sketch's Euler outputs indicate true north. Looking at the code that makes more sense, so perhaps I measured it backwards at the time or just wrote it up backwards. I didn't have a compass handy to check, unfortunately. Either way, the Euler and quaternion sketch outputs are not measuring the same thing, and that wasn't obvious to me.

closed 2020-12-21 by gregtomasch