gregtomasch/EM7180_SENtral_Calibration #9: AlgorithmControl set to 0x00 twice?

gerowtj, 2019-09-25:
In the EM7180_SENtral_Calibration/Teensy_ST_USFS_Baseline_Calibration_Utility_Acc_WS/EM7180.cpp
program, lines 193 to 214: 

```
// ODR + 10000000b to activate the eventStatus bit for the barometer...
  I2C->writeByte(EM7180_ADDRESS, EM7180_BaroRate, (0x80 + BARO_ODR));

  // Configure operating mode
  // Output scaled sensor data (Quaternion convention NED)
  **I2C->writeByte(EM7180_ADDRESS, EM7180_AlgorithmControl, 0x00);**
  
  // Enable interrupt to host upon certain events
  // Choose interrupts when: gyros updated (0x20), Sentral error (0x02) or Sentral reset (0x01)
  I2C->writeByte(EM7180_ADDRESS, EM7180_EnableEvents, 0x23);

  #ifdef SERIAL_DEBUG
    Serial.print("Done. Starting the Sentral...");
    Serial.println("");
  #endif

  // Start the Sentral
  **I2C->writeByte(EM7180_ADDRESS, EM7180_AlgorithmControl, 0x00);**
  #ifdef SERIAL_DEBUG
    Serial.print("Done. Loading algorithm tuning parameters...");
    Serial.println("");
  #endif
```
the `I2C->writeByte(EM7180_ADDRESS, EM7180_AlgorithmControl, 0x00);` instruction is done twice, once after setting the BaroRate and then again after writing to the EnableEvents register.

Is this required?  Or can the write of 0x00 to the AlgorithmControl register be done once?

Also, the write to EnableEvents = 0x23 is done on line 202 and line 228, Is there a reason for this as well?

Thank you for your time!
Tom

gregtomasch, 2019-10-07:
Setting "EM7180_AlgorithmControl" to 0x00 re-starts SpacePoint the algorithm. The second call certainly isn't hurting anything. If you find the second call offensive, comment it out and see if it causes any malfunction...

gerowtj, 2020-11-12:
Commenting out the line of code did not cause any issues, that I have detected.

closed 2020-11-12 by gerowtj