gregtomasch/EM7180_SENtral_Calibration #11: Can't perform the callibration on a Onehorse ESP32 dev board

Zieju, 2019-11-04:
Dear Greg
I have an issue with your ESP32_ST_USFS_Baseline_Calibration_Utility_Acc_WS sketch.
A serial monitor freezes imediatelly after sending "1" to capture calibration values. It happens in any direction (not only in X pointing +1G). I tried to make the calibration on two different USFS boards (one brand new and the second tested before on teensy) and the result is the same. The connection is good.
On teensy 3.2 the calibrations works well. Could you help me please?
Kamil

Zieju, 2019-11-04:
After some research I found the reason of this strange behaviour. I had very old arduino core for esp32 and this caused problems (perhaps with i2c communication with eeprom). After the update to 1.04 version succesfully performed the callibration. I hope it will help to someone :) Cheers

closed 2019-11-04 by Zieju

gregtomasch, 2019-11-05:
Thanks for the info; all's well that ends well...